dotchen/LearningByCheating #1: Number of pedestrians in dense No-Crash benchmark Town1/Town2

marintoro, 2019-11-16:
Hello,

Really nice code and re-implementation of the CARLA benchmark!
I found something weird, in the benchmark/__init__.py where the scenarios are defined, it's written that the NoCrashTown01-v5 scenario got 250 pedestrians while the NoCrashTown02-v5 has only 150. However, I launched both scenario and it seems like there is much much more pedestrians spawned and crossing roads in NoCrashTown02-v5. 
Is that a bug? Or maybe I didn't understand something there...

dianchen96, 2019-11-18:
Thank you for your interest in our project. You could take a look at the logic for spawning pedestrians here: https://github.com/dianchen96/LearningByCheating/blob/release-0.9.6/bird_view/utils/carla_utils.py#L422. Note that Town2 is smaller than Town1 and much smaller walkable region defined in the navmesh, which results in a higher pedestrian density at walkable regions. Please let us know if you still believe there is a bug.

marintoro, 2019-11-18:
Thanks for this fast reply. I think the _Town2 is smaller than Town1 and has much smaller walkable region_ totally answer my question!
As a results my agent is way worse in _Town2 dense_ scenario than _Town1 dense_ scenario even if I actually trained it on Town2.

closed 2020-01-07 by dianchen96