dotchen/LearningByCheating #3: Get `Vehicle' object has no attribute 'stop_dtcrowd'` when running benchmark

jrabary, 2020-01-15:
Hi,
Thanks for sharing your benchmark code which is really helpful.
We are trying to use it to evaluate our agent but we get the following error:

```
'Vehicle' object has no attribute 'stop_dtcrowd'
```

This appears when we run the example code described in the readme. Is there something we missed ?

jrabary, 2020-01-15:
Solve the problem by installing the specific `.egg` file. I miss the line describing it on the readme.

closed 2020-01-15 by jrabary

Chuqt, 2025-02-23:
Hello, may I ask the website provided in the tutorial to download egg, "http://www.cs.utexas.edu/~dchen/lbc_release/egg/carla-0.9.6-py3.5-linux-x86_64.egg", I have tried for a long time can not open, may I ask you how to solve it, thank you very much