dotchen/LearningByCheating #6: phase 2 running out of memory

ptigas, 2020-01-31:
Hi,

Thank you for releasing the code, this is great!

I'm running phase2 however it runs out of memory after 3-4 epochs and the problem seem to come from replay buffer, which is always growing. My server has 52GB and 10GB swap which should have been enough (for 200,000 frames which is the maximum buffer). Did you face similar issues? If not, how much RAM did you use?

p.

dianchen96, 2020-01-31:
Thanks for your interest in our project!

Each frame consists of a 192x192x7 uint8 map and a 384x160x3 uint8 rgb image, so it takes up a lot of space when it is full. The server I use has 256GB RAM so I never experienced this. 

I would recommend reducing the buffer size and see if it affect the performance. If so, try saving the trajectory data to disk and load them for training.

ptigas, 2020-02-08:
Thank you for your response! Unfortunately 256GB RAM is not an option for me. I'm currently running it with swap of suggested RAM size and then I'll try your suggestions. 

I'll report the results here.

p.

ptigas, 2020-02-16:
Seems to be working. I can repro the 100% on town01 after dagger with swap memory scaled up.

The following commands do the trick
```
sudo fallocate --length 150GiB /swapfile
sudo mkswap /swapfile
sudo swapon /swapfile
```

closed 2020-02-16 by ptigas