dotchen/LearningByCheating #9: benchmark_agent.py fails due to missing `agents` module

t27, 2020-03-02:
When we directly run the `benchmark_agent.py` script after following the instructions, we get this error. (Using this command `CUDA_VISIBLE_DEVICES="0" python benchmark_agent.py --suite=town2 --model-path=ckpts/image/model-10.th --show`)

```
Traceback (most recent call last):
  File "benchmark_agent.py", line 6, in <module>
    from benchmark import make_suite, get_suites, ALL_SUITES
  File "/home/user/carla/lbc/carla_lbc/benchmark/__init__.py", line 1, in <module>
    from .goal_suite import PointGoalSuite
  File "/home/user/carla/lbc/carla_lbc/benchmark/goal_suite.py", line 9, in <module>
    from agents.navigation.local_planner import RoadOption, LocalPlannerNew, LocalPlannerOld
ImportError: No module named 'agents'
```

This is because python tries to find the `agents` module in the carla package's path(this is in the `site-packages` folder inside the `carla` conda environment).

I had to copy the `PythonAPI/agents` folder alongside the `carla` package folder in the site-packages for it to run properly.

I ran these commands to fix this,(tried to make the commands generic for any python env setup)
```
# ensure you have activated the carla conda environment
export CARLALIB=`python -c "import carla;print(carla.__path__[0])"`
# ~/anaconda3/envs/carla/lib/python3.5/site-packages/carla-0.9.6-py3.5-linux-x86_64.egg/carla

# need to copy the PythonAPI/agents folder to the installed carla package
# ensure you are in the carla_lbc directory that we created in the initial setup
cp -r PythonAPI/agents $CARLALIB/../
```

I faced this when following the instructions in the README, is there a better fix for this? 
I can add the above commands to README in a PR as well

dianchen96, 2020-03-03:
I believe this is already covered by README. See https://github.com/dianchen96/LearningByCheating#data-collection

t27, 2020-03-03:
Yeah, updating the PYTHONPATH would solve it as well, we should also add that to the QuickStart section of README where the inference command of the benchmark_agent is mentioned.

Will add a PR for this soon!

closed 2020-03-04 by t27

AIasd, 2020-04-12:
In my case, 
just updating the PYTHONPATH does not work but running these commands you mentioned fix the issue:
```
# ensure you have activated the carla conda environment
export CARLALIB=`python -c "import carla;print(carla.__path__[0])"`
# ~/anaconda3/envs/carla/lib/python3.5/site-packages/carla-0.9.6-py3.5-linux-x86_64.egg/carla

# need to copy the PythonAPI/agents folder to the installed carla package
# ensure you are in the carla_lbc directory that we created in the initial setup
cp -r PythonAPI/agents $CARLALIB/../
```