dotchen/LearningByCheating #12: Extending this repo for Town03.

deval-maker, 2020-03-24:
Hello, 
I am trying to run the agent (pre-trained) in the Town03. I am getting the following error.

```
Traceback (most recent call last):
  File "benchmark_agent.py", line 85, in <module>
    run(Path(args.model_path), args.port, args.suite, args.big_cam, args.seed, args.autopilot, args.resume, max_run=args.max_run, show=args.show)
  File "benchmark_agent.py", line 61, in run
    run_benchmark(agent_maker, env, benchmark_dir, seed, autopilot, resume, max_run=max_run, show=show)
  File "/home/deval/Code/carla/carla_lbc/benchmark/run_benchmark.py", line 241, in run_benchmark
    result, diagnostics = run_single(env, weather, start, target, agent_maker, seed, autopilot, show=show)
  File "/home/deval/Code/carla/carla_lbc/benchmark/run_benchmark.py", line 169, in run_single
    env.init(start=start, target=target, weather=cu.PRESET_WEATHERS[weather])
  File "/home/deval/Code/carla/carla_lbc/benchmark/goal_suite.py", line 52, in init
    super().init(**kwargs)
  File "/home/deval/Code/carla/carla_lbc/bird_view/utils/carla_utils.py", line 509, in init
    map_utils.init(self._client, self._world, self._map, self._player)
  File "/home/deval/Code/carla/carla_lbc/bird_view/utils/map_utils.py", line 1338, in init
    module_manager.start_modules()
  File "/home/deval/Code/carla/carla_lbc/bird_view/utils/map_utils.py", line 147, in start_modules
    module.start()
  File "/home/deval/Code/carla/carla_lbc/bird_view/utils/map_utils.py", line 611, in start
    self.map_image = MapImage(self.world, self.town_map, PIXELS_PER_METER)
  File "/home/deval/Code/carla/carla_lbc/bird_view/utils/map_utils.py", line 388, in __init__
    carla_world, carla_map, self.world_to_pixel, self.world_to_pixel_width)
  File "/home/deval/Code/carla/carla_lbc/bird_view/utils/map_utils.py", line 482, in draw_road_map
    does_cross_solid_line(sample, -sample.lane_width * 1.1))
  File "/home/deval/Code/carla/carla_lbc/bird_view/utils/map_utils.py", line 400, in draw_lane_marking
    pygame.draw.lines(surface, COLOR_WHITE, False, points, 2)
ValueError: points argument must contain more than 1 points

```

Any help on this would be appreciated.

Thanks!

dianchen96, 2020-03-24:
Hi,

This repo currently only supports Town1 and Town2. To get around the error I think you can add 

```
if len(points) <= 1:
    continue
```

before the draw lane marking code.

Regardless, the pre-trained model will not work on Town3 because it is very different from Town1 and Town2.

lakshin, 2021-01-01:
@deval-maker I managed to get the pre-trained model work with Town3 as well as other available towns. I got it to work by following the below steps. 

1) Locate __init__.py inside the benchmark folder and add a new elif Town03 to the _add function.
2) In __init__.py you will have to call the _add function and register Town03
`_add('StraightTown03-v1', n_vehicles=20, n_pedestrians=20, weathers=WEATHER_1)` 
3) In __init__.py locate the _aliases array and a new alias for town3
`_aliases = {
        ...
        'town3': ['StraightTown03-v1'],
        ...
            }`
4) Navigate to benchmark -> corl2017 -> 096 and create a new file named 'straight_Town03.txt' 
5) Inside the newly created file add start and target indexes as shown below in the screenshot (Indexes used get a spawn point from the get_spawn_points() function)
[image]

6) Finally run - 'python benchmark_agent.py --suite=town3 --model-path=ckpts/image/model-10.th --show'

As @dotchen mentioned seems like the pre-trained model doesn't work well in town3 as it contains multiple lanes (as shown in the below video). This awesome project could be extended and improved to work in multiple lane scenarios like town3. 

https://user-images.githubusercontent.com/9911515/103444948-27b55c80-4c94-11eb-8a68-624db3ab2b51.mp4

Enjoy!!

dotchen, 2021-05-05:
If you would like a baseline that drives in Town03, please checkout our follow-up work! https://github.com/dotchen/WorldOnRails

closed 2021-05-05 by dotchen