dotchen/LearningByCheating #15: Issue when cloning this repo over the extracted Carla tarball

t27, 2020-04-12:
In the instructions mentioned in the README and in the quickstart.sh, we first extract the Carla tarball and then clone this repository over that. 

When we do this,
```
# Download LBC
git init
git remote add origin https://github.com/dianchen96/LearningByCheating.git
git pull origin release-0.9.6
```
After the last line, git aborts the pull process as there is a conflict in the LICENSE file. (the Carla zip also includes a LICENSE file with the same name)

The current workaround that I use is to add a simple rename command after the tar extraction and before `git init` like this

```
mv LICENSE LICENSE_CARLA
```

After this the git pull seems to work properly.

dianchen96, 2020-04-12:
Thanks for poinint this out! I made the instructions before adding the license file and I forgot to address that. Do you mind creating a PR with your fix above?

t27, 2020-04-13:
Sure, I'll update the readme and quickstart files

closed 2020-07-04 by dianchen96