dotchen/LearningByCheating #16: fail to run benchmark_agent.py

AIasd, 2020-04-13:
Dear Authors,

I followed the quick start script and then run `./CarlaUE4.sh -fps=10 -benchmark` and in a separate tab run `CUDA_VISIBLE_DEVICES="0" python benchmark_agent.py --suite=town2 --model-path=ckpts/image/model-10.th --show`.

The carla gets run successfully (except the mouse gets captured inside the window opened by carla).
However, for the benchmark_agent script, after running it, only the following two lines show up in the terminal and it gets stuck here forever:
```
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
```

Could you shed some lights on this misbehavior? Thank you so much!

dianchen96, 2020-04-13:
Check if it is related to #7 ?

AIasd, 2020-04-14:
Yes, this solved the problem. I did not notice that. Thanks a lot!

closed 2020-04-14 by AIasd