dotchen/LearningByCheating #23: environment.yml with different cuda and cudnn versions

abdallahaymaan, 2020-07-04:
Hello, I would like to know what should be changed in the environment.yml files if I installed different cuda and cudnn versions since cuda version 8.0 isnt compatible with ubuntu 18.04

- CUDA v.10.1.103
-Cudnn v.7.6.5.32
-Tensorflow-gpu v.2.2

**If anyone went through this and can help please do, Thanks in advance**

dianchen96, 2020-07-04:
Install the later pytorch version that supports later cuda should be okay

abdallahaymaan, 2020-07-04:
I get this error when running the line

> conda env create -f environment.yml
after I have removed lines concerning cuda, cudnn, pytorch, and tensorflow
# >>>>>>>>>>>>>>>>>>>>>> ERROR REPORT <<<<<<<<<<<<<<<<<<<<<<

    Traceback (most recent call last):
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/exceptions.py", line 1079, in __call__
        return func(*args, **kwargs)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda_env/cli/main.py", line 80, in do_call
        exit_code = getattr(module, func_name)(args, parser)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda_env/cli/main_create.py", line 111, in execute
        result[installer_type] = installer.install(prefix, pkg_specs, args, env)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda_env/installers/conda.py", line 32, in install
        prune=getattr(args, 'prune', False), update_modifier=UpdateModifier.FREEZE_INSTALLED)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/core/solve.py", line 117, in solve_for_transaction
        should_retry_solve)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/core/solve.py", line 158, in solve_for_diff
        force_remove, should_retry_solve)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/core/solve.py", line 281, in solve_final_state
        ssc = self._run_sat(ssc)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/common/io.py", line 88, in decorated
        return f(*args, **kwds)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/core/solve.py", line 808, in _run_sat
        should_retry_solve=ssc.should_retry_solve
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/common/io.py", line 88, in decorated
        return f(*args, **kwds)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/resolve.py", line 1318, in solve
        self.find_conflicts(specs, specs_to_add, history_specs)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/resolve.py", line 347, in find_conflicts
        bad_deps = self.build_conflict_map(specs, specs_to_add, history_specs)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/resolve.py", line 507, in build_conflict_map
        root, search_node, dep_graph, num_occurances)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/resolve.py", line 369, in breadth_first_search_for_dep_graph
        last_spec = MatchSpec.union((path[-1], target_paths[-1][-1]))[0]
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/models/match_spec.py", line 481, in union
        return cls.merge(match_specs, union=True)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/models/match_spec.py", line 475, in merge
        reduce(lambda x, y: x._merge(y, union), group) if len(group) > 1 else group[0]
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/models/match_spec.py", line 475, in <lambda>
        reduce(lambda x, y: x._merge(y, union), group) if len(group) > 1 else group[0]
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/models/match_spec.py", line 502, in _merge
        final = this_component.union(that_component)
      File "/home/abdallahaymaan/anaconda3/lib/python3.7/site-packages/conda/models/match_spec.py", line 764, in union
        return '|'.join(options)
    TypeError: sequence item 0: expected str instance, Channel found

`$ /home/abdallahaymaan/anaconda3/bin/conda-env create -f environment.yml python=3.5`

  environment variables:
                 CIO_TEST=<not set>
  CONDA_AUTO_UPDATE_CONDA=false
        CONDA_DEFAULT_ENV=base
                CONDA_EXE=/home/abdallahaymaan/anaconda3/bin/conda
             CONDA_PREFIX=/home/abdallahaymaan/anaconda3
    CONDA_PROMPT_MODIFIER=(base)
         CONDA_PYTHON_EXE=/home/abdallahaymaan/anaconda3/bin/python
               CONDA_ROOT=/home/abdallahaymaan/anaconda3
              CONDA_SHLVL=1
                     PATH=/home/abdallahaymaan/anaconda3/bin:/usr/local/cuda-10.1/bin:/usr/local
                          /cuda-10.1/NsightCompute-2019.1:/home/abdallahaymaan/anaconda3/bin:/ho
                          me/abdallahaymaan/anaconda3/condabin:/home/abdallahaymaan/.local/bin:/
                          usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games
                          :/usr/local/games:/snap/bin
       REQUESTS_CA_BUNDLE=<not set>
            SSL_CERT_FILE=<not set>
               WINDOWPATH=1

     active environment : base
    active env location : /home/abdallahaymaan/anaconda3
[... 6 lines cut ...]
       virtual packages : __cuda=10.2
                          __glibc=2.27
       base environment : /home/abdallahaymaan/anaconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/abdallahaymaan/anaconda3/pkgs
                          /home/abdallahaymaan/.conda/pkgs
       envs directories : /home/abdallahaymaan/anaconda3/envs
[... cut ...]
               platform : linux-64
             user-agent : conda/4.8.2 requests/2.20.1 CPython/3.7.6 Linux/5.3.0-62-generic ubuntu/18.04.4 glibc/2.27
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False

An unexpected error has occurred. Conda has prepared the above report.

If submitted, this report will be used by core maintainers to improve
future releases of conda.
Would you like conda to send this report to the core maintainers?

[y/N]: 
Timeout reached. No report sent.

closed 2020-07-06 by abdallahaymaan

shirgoldfinger, 2022-10-25:
@abdallahaymaan Would you mind sharing the .yaml file you ended up using to get the environment to build?

xolovezari, 2024-08-05:
> @abdallahaymaan Would you mind sharing the .yaml file you ended up using to get the environment to build?

Hey! Did you find out the .yaml file?