dotchen/LearningByCheating #24: Error in Training teacher model

FitashUlHaq, 2020-07-04:
I have generated 100 episodes using the given script but when i try to train the birdview i am getting this error

MDB_INVALID: File is not an LMDB file

can you please guide me what can be the issue?

any help will be appreciated.
Many thanks
Fitash

dianchen96, 2020-07-04:
Make sure you pass the correct path to data when running the training script.

FitashUlHaq, 2020-07-04:
the path is correct with two folders train and val both containing different episodes

dianchen96, 2020-07-04:
Can you print the fullpath here: https://github.com/dianchen96/LearningByCheating/blob/release-0.9.6/bird_view/utils/datasets/birdview_lmdb.py#L64 and see if it makes sense?

If yes then check the dataset size. If both make sense then the issue might be related to the lmdb installation.

dianchen96, 2020-07-05:
Closing it for now. If this is not solved feel free to reopen it and commen below.

closed 2020-07-05 by dianchen96

FitashUlHaq, 2020-07-06:
I tried:

- Reinstalling lmdb
- ignoring the bad files
- regenerating episodes

only few episodes can be used, all others gave the same error 

I am using ubunutu 18.04, python 3.6.9, lmdb 0.94

dianchen96, 2020-07-06:
Can you try running in the python environment we provided in the setup instructions? That uses python 3.5.

FitashUlHaq, 2020-07-06:
I tried that first but ended up with this results
UnsatisfiableError: The following specifications were found to be in conflict:
  - blas 1.0 mkl
  - gmp 6.1.2 h6c8ec71_1 -> libgcc-ng 7.2.0.*
  - gmp 6.1.2 h6c8ec71_1 -> libstdcxx-ng 7.2.0.*
Use "conda info <package>" to see the dependencies for each package.

FitashUlHaq, 2020-07-06:
I tried that and same error as 

/media/fitash/Carladata/train/001: MDB_INVALID: File is not an LMDB file

FitashUlHaq, 2020-07-06:
which ubuntu version are you using?

I am using 18.04

dianchen96, 2020-07-06:
Thanks for the details. I used 18.04 and 16.04 and both worked for me. Do the corrupt runs have the correct files in it (i.e lock.mdb etc.) & have the correct file size?

FitashUlHaq, 2020-07-06:
the lock files are of the same size while the data file is sometime 2.0gbs sometime 1.8 and 2.2. Mostly the corrupt files are other than the last 6 files

dianchen96, 2020-07-06:
This is odd... the file sizes look correct to me. I will try your env and see if i can reproduce it. In the mean time can you try what if you just collect 1 episode and see if it is corrupted. Also maybe take a a look at https://stackoverflow.com/questions/44407659/how-to-force-64-bit-lmdb-to-generate-a-32-bit-database if your machine is 32-bit.