dotchen/LearningByCheating #27: Tips on efficient evaluation

peiyunh, 2020-07-20:
Hi Dian, 

I am trying to reproduce the NoCrash benchmark results. It takes about 15 hours to do 100 runs for the empty suite and about 21 hours for the dense suite. I wondering if those numbers are in the ballpark or if there is a much more efficient way go to about it. 

I would kick off 3 CARLA instances per GPU through docker. Below is an example, 
```
docker run -p 2000-2002:2000-2002 --gpus '"device=0"' carlasim/carla:0.9.6 ./CarlaUE4.sh -fps 10 -benchmark -world-port='2000'
```

Then I'd run the benchmark code for 3 agents per GPU as well. Below is an example, 
```
PYTHONPATH='$HOME/software/CARLA_0.9.6/PythonAPI:$HOME/code/lbc/PythonAPI' CUDA_VISIBLE_DEVICES=0 python benchmark_agent.py --model-path ckpts/image/model-10.th --port 2000 --suite empty --seed 1 --max-run 100
```

In terms of speeding up the evaluation, I was thinking about cranking up the FPS of CARLA, but I'm not sure at all if that's recommended. I'd appreciate it if you could offer some suggestions. 

Thanks!

dianchen96, 2020-07-20:
Yes, this is unfortunately the case for me as well as the suite is very big :(

I got away by paralizing the jobs on the cluster. If you have a similar setting and think this can help, I can send you my slurm jobs scripts.

peiyunh, 2020-07-21:
I am running it locally but it is great to know that I am not doing it wrong. Thanks a lot!

closed 2020-07-21 by peiyunh