dotchen/LearningByCheating #28: Weird cases where autopilot would ignore the red light

peiyunh, 2020-08-05:
I found some weird cases where autopilot would ignore the red light, which was also mentioned in another thread: https://github.com/dianchen96/LearningByCheating/issues/2#issuecomment-668961125. 

I am posting it here in case others are curious. Below are some videos. Interestingly, it looks like all cases happen at the same intersection. I am wondering if this is the one traffic light @dianchen96 mentioned about. 

[image]

[image]

[image]

dianchen96, 2020-08-06:
Were you able to reproduce the numbers though? @peiyunh

peiyunh, 2020-08-06:
Yeah! I was able to reproduce the autopilot's numbers. These are just cases that I find intriguing. 

Great suggestion. I will test later versions when I get a chance. Will close the issue for now. 

Thanks.

closed 2020-08-06 by peiyunh