dotchen/LearningByCheating #29: Is there a way to check if I installed the egg file correctly?

peiyunh, 2020-08-13:
I was browsing the visualizations you shared via Google drive. I noticed pedestrians cross the streets quite often. It struck me that I have never seen this happening when I run simulation locally. I suspect it could be that I did not install the egg file correctly. I know you have mentioned fixes you've done regarding pedestrians. I wonder if there is a way to check if I have installed the patched version correctly. 

Thanks a lot. I really appreciate the help!

dianchen96, 2020-08-13:
If none of the pedestrians are crossing my guess is that the navmesh file is not setup correctly? 

Make sure you have http://www.cs.utexas.edu/~dchen/lbc_release/navmesh/Town01.bin and http://www.cs.utexas.edu/~dchen/lbc_release/navmesh/Town01.bin downloaded and replacing the original files.

peiyunh, 2020-08-13:
Great catch! I don't know how I missed that, but pedestrians now are able to cross streets after I replaced the navmesh file. Thanks!!

peiyunh, 2020-08-13:
Btw, if I want to run this offscreen via docker, I will have to rebuild the docker image, right?

dianchen96, 2020-08-13:
No it should work out of the box. The CARLA docker is independent of the CARLA binary.

closed 2020-08-13 by dianchen96

peiyunh, 2020-08-13:
I was using the official docker image for CARLA 0.9.6, which has everything (including navmesh files) bundled and is immutable. It turns out I have to build a docker image after replacing those navmesh files and run the new docker image to see pedestrians crossing. Thanks for the help!

dianchen96, 2020-08-13:
You shouldn't need to because I was using the docker image as well on the cluster. Inside docker image you can still run the modified CARLA.