dotchen/LearningByCheating #33: Found conflicts error on creating environment.

ddroder, 2020-10-19:
Hello! I am following the installation from the readme that you have provided and am stuck on getting the environment created! Whenever I create the environment, it ends up saying it found conflicts. Here is a screenshot of what I get!

[image]

It will run this for several minutes, and then crash with a massive output of package conflicts. Any advice would be awesome, thanks!

ddroder, 2020-10-19:
Managed to fix that error by moving that line down to the pip section. Now I am getting a "Found conflicts! Looking for incompatible packages" message. Here is a screenshot:
[image]

dianchen96, 2020-10-19:
Hello! Is this error caused by installing Pytorch? If so I recommend removing it and manually install it later using conda install xxx

ddroder, 2020-10-19:
Thanks so much for the response!
Unfortunately, it is not the installation of PyTorch that is the issue! When I comment out all of the installations for PyTorch from the yml file, I still receive this error! For context, I am just running the quick_start.sh file which will create the new environment for me.

RifkyYP, 2020-10-20:
> Hello! I am following the installation from the readme that you have provided and am stuck on getting the environment created! Whenever I create the environment, it says ResolvePackageNotFound and points to cudatoolkit. Here is a screenshot of what I get!
> [image]
> 
> Any advice on this would be awesome. Thanks so much!

I'm also getting this issue when creating the environment. My question is, can i just install the newer version of cudatoolkit package? is there any effect to the compatibility of your program? @dianchen96

Rookie-81, 2020-12-15:
getting same error while configuring the conda env

shirgoldfinger, 2022-10-25:
I am getting the same issue while configuring the conda environment.