dotchen/LearningByCheating #36: RPC connection error

Rookie-81, 2020-12-16:
Hello everyone when i ever i try to execute `CUDA_VISIBLE_DEVICES="0" python benchmark_agent.py --suite=town2 --model-path=ckpts/image/model-10.th --show`

this error shows up

world = client.get_world()
RuntimeError: rpc::rpc_error during call in function version

any help will be appreciated.

Rookie-81, 2020-12-16:
Closing because i've found the solution
1)i did pip3 uninstall carla
2) had to find a workaround to get easy_install to work and installed the provided egg
Thank you, and awesome work!

closed 2020-12-16 by Rookie-81

Kin-Zhang, 2021-06-09:
[image]
I think I maybe have the same problem but different error or can say that the client cannot get the world.
But I tried the 
uninstall and install provided egg ... still cannot work.

Kin-Zhang, 2021-06-10:
> [image]
> I think I maybe have the same problem but different error or can say that the client cannot get the world.
> But I tried the
> uninstall and install provided egg ... still cannot work.

OK I finally got the way, you cannot run the benchmark_agent first, when I run other file like data_collection.py it's okay. Just for other people who may meet this question.