dotchen/LearningByCheating #37: data_collector.py saves files data.mdb and lock.mdb

Jidenna, 2021-03-16:
I tried runnung 

`python data_collector.py --dataset_path=dataset`

but the output I get on the dataset folder ends up being a data.mdb and lock.mdb file rather than the .lmdb in the train and val folder as highlighted on your README.md. What am I doing wrong? 

How do I get the files to be a .lmdb file type in a train and val folder needed to train the privileged agent?

dotchen, 2021-05-05:
lmdb files are saved as the lock and data file as the format.

closed 2021-05-05 by dotchen

Jidenna, 2021-09-09:
What does that mean? 

After runnning the data_collector.py script successful. I tried to run the cd training
python train_image_phase0.py --dataset_path=[DATA PATH] --log_dir=[LOG DIR] --pretrained --teacher_path=[TEACHER PATH]