dotchen/LearningByCheating #38: Which version of Ubuntu does this repo work with?

Shubhamcl, 2021-03-28:
I am asking as with Ubuntu 20 and 18 it the environment does not built, and complains about not having cudatoolkit 8.
While cudatoolkit 8 is only compatible with 16 and 14 as per the Nvidia website.

Would that mean this repository is still stuck on Ubuntu 16 or maybe even 14?

dotchen, 2021-03-28:
I think the repo works with later versions of cuda/pytorch. Maybe try removing the lines in the yaml file and manually install them later.

Shubhamcl, 2021-04-08:
That seemed to help!

Basically starting with building the environment is quite demotivating because it ends up in many errors. A better way to start is with a base environment and some basic packages preinstalled then eventually installing the missing dependencies by yourself manually to fix the errors when you straight away execute the code.

Versions can be matched by looking at the dependencies file. To sum it up the list of dependencies might be too long, just install what is need at the moment one by one.

closed 2021-04-08 by Shubhamcl

shirgoldfinger, 2022-10-25:
@Shubhamcl Would you mind saying exactly which lines you commented out/removed from the yaml file? I'm trying to build the environment and it's still not working.