dotchen/LearningByCheating #39: I have some questions about the CoordConverter

raozhongyu, 2021-04-13:
Thanks for your great work, I have some questions about the CoordConverter in train_image1. I'd like to konw that if i change the image size to (192,192), shall I change the world_y and fixed_offset

dotchen, 2021-04-14:
No need to change world_y, just change the corresponding image height and width in `CoordConverter`. You might want to tune fixed_offset though since the default value was tuned for 384x160

raozhongyu, 2021-04-14:
Thanks for your reply, how to change the fixed_offset. Is there a formula

dotchen, 2021-04-15:
`fixed_offset` is used to offset predictions that are too close so that they still lie in the image when projected. There is no formula for the optimal `fixed_offset`, but you can look at the pinhole camera math to see what values it correspond in the image space.

raozhongyu, 2021-04-15:
Thanks a lot

closed 2021-05-05 by dotchen