dotchen/LearningByCheating #41: Unable to reproduce test town, test weather results

HimangiM, 2021-05-05:
Hi, 

I am trying to reproduce the results on noCrash benchmark using the pre-trained model, model-10.ckpt and my results are as follows:

> Suite Name | Seed 0 | Seed 1 | Seed 2 | Our implementation (Avg of seed 0, 1, 2) | LBC (Seed 0, 1, 2)
> -- | -- | -- | -- | -- | --
> NoCrashTown02-v1 | 99 | 99 | 99 | 99 | 99.0 土 0.0
> NoCrashTown02-v2 | 68 | 66 | 66 | 64 | 84.7 土 3.1
> NoCrashTown02-v3 | 95 | 97 | 91 | 94.3 | 93.3 土 2.1
> NoCrashTown02-v4 | 66 | 58 | 54 | 59.3 | 70.0 土 2.0
> NoCrashTown02-v5 | 47 | 52 | 44 | 47.6 | 46.1 土 4.0
> NoCrashTown02-v6 | 24 | 30 | 32 | 28.6 | 32.7 土 9.5

I am able to reproduce the results for test town and train weathers (v1, v3, v5), however, for the other 3 conditions with test town and test weathers (v2, v4, v6), there is a significant decrease from the numbers reported by LBC. Can you please let me know what can be the issue?

Thanks,
Himangi

dotchen, 2021-05-05:
I am a bit confused. What does `Our implementation' mean? Also, the mean of 68,66,66 does not seem to be 64 (row 2 above).

Anyway, I'd suggest making sure you have not changed the implementation of this repo. Here is a [reference](https://github.com/dotchen/LearningByCheating/commit/a04e94afad6346f56a17c40007a8fbd4e5647325#comments) where somebody else has reproduced the numbers on NoCrash.

dotchen, 2021-05-05:
Also, if you are comparing against LBC as a baseline in your research, please consider also checking out our latest work: https://github.com/dotchen/WorldOnRails, which outperforms LBC and ranks first on the public leaderboard.

closed 2021-05-05 by dotchen